untr0py/SevSeg #2: undefined reference to operator new unsigned int

NicAshby, 2015-01-23:
This is what happened when I tried to compile the example.
Am I missing something?

Arduino: 1.5.8 (Windows 7), Board: "Digispark Pro (Default 16 Mhz)"  SevSeg\SevSeg.cpp.o: In function `SevSeg::begin(unsigned char, unsigned char, unsigned char*, unsigned char*)': C:\Program Files (x86)\Digistump_Arduino\libraries\SevSeg/SevSeg.cpp:74: undefined reference to `operator new[](unsigned int)' C:\Program Files (x86)\Digistump_Arduino\libraries\SevSeg/SevSeg.cpp:75: undefined reference to `operator new[](unsigned int)' collect2: error: ld returned 1 exit status Error compiling.    This report would have more information with   "Show verbose output during compilation"   enabled in File > Preferences.

DeanIsMe, 2015-01-25:
Hmm, I just compiled it and got the same warning. Although, I don't remember this happening previously. I'll have a look in a couple days when I get the chance.

DeanIsMe, 2015-02-02:
Fixed. Avoided use of operator new. See commit on 02/02/2015

closed 2015-02-02 by DeanIsMe