untr0py/SevSeg #7: Please issue a release

leres, 2016-09-17:
Could you please issue a release with a version number? I have a FreeBSD port I'd like to submit and although it's possible to download the current version using the tag name ba10676f8505b5ceb47d32f6ef3222ef2067c4cd, having an actual release version number would be much cleaner.

Thanks!

DeanIsMe, 2016-09-18:
Release issued with version number 3.1.

closed 2016-09-18 by DeanIsMe

leres, 2016-09-18:
Thanks for the quick response. However it looks like you put a version 3.1 comment in one of the source files. What I was hoping for was a github release called 3.1. That way the FreeBSD port can download a package that contains all of the files as of the release date.

DeanIsMe, 2016-09-18:
Ah I see. I wasn't aware of the 'Release' option on Github. I believe that I have now made v3.1 the first 'Release'.

leres, 2016-10-13:
On 09/17/16 18:01, Dean Reading wrote:
> Ah I see. I wasn't aware of the 'Release' option on Github. I believe
> that I have now made v3.1 the first 'Release'.

FYI, this is finally available as a FreeBSD port:

    https://www.freshports.org/devel/arduino-sevseg/

		Craig