untr0py/SevSeg #10: Not an Issue, Need help(first post.sorry for protocol breach)

saurabhreigns, 2016-11-24:
Please help me understand why this code does not work. 
[code]
//Library
#include "SevSeg.h"
#include<CountUpDownTimer.h>
#include <Button.h>

//#define setPin  0 
//#define plusPin 1
//#define minusPin 2
//#define startTimerPin 3
//create a Button objects at pin a0-a3
// Uses debounce mode. You may need to experiment with the debounce duration
// for your particular switch.
#define ledPin A4

Button setPin = Button(A0, BUTTON_PULLUP_INTERNAL, true, 10);
Button plusPin = Button(A1, BUTTON_PULLUP_INTERNAL, true, 10);
Button minusPin = Button(A2, BUTTON_PULLUP_INTERNAL, true, 10);
Button startTimerPin = Button(A3, BUTTON_PULLUP_INTERNAL, true, 10);

int seconds=0000;

CountUpDownTimer T(DOWN);

SevSeg sevseg; //Instantiate a seven segment controller object

//Variables
  byte secvalue =0;
  byte milvalue =0;

void setup() {

  byte numDigits = 4;   
  byte digitPins[] = {5,4,3,2,1};
  byte segmentPins[] = {6, 7, 8, 9, 10, 11, 12, 13};

  pinMode(ledPin,OUTPUT);

  sevseg.begin(COMMON_ANODE, numDigits, digitPins, segmentPins);
  sevseg.setBrightness(50);
  //sevseg.setNumber(1234,0);
  //while(1){sevseg.refreshDisplay();};
      digitalWrite(ledPin,HIGH);//indicator LED test
      delay(500);
      digitalWrite(ledPin,LOW);
}

void loop(){
    //sevseg.setNumber(seconds,0);

    if(setPin.uniquePress())
    {
      seconds=0;//resetting the timer value
      sevseg.setNumber(seconds,0);
    }
          
    else if(plusPin.uniquePress())
    {
      seconds=seconds++;
      sevseg.setNumber(seconds,0);
    }
    
    else if(minusPin.uniquePress())
    {
      seconds=seconds--;
      sevseg.setNumber(seconds,0);
    }
     
    else if(startTimerPin.uniquePress())
    {
      T.SetTimer(seconds);
      T.StartTimer();
      secvalue=T.ShowSeconds();
      milvalue=T.ShowMilliSeconds();
      sevseg.setNumber(seconds,0);
    }
      sevseg.setNumber(seconds,0);
      sevseg.refreshDisplay();
}
[/code]

DeanIsMe, 2016-11-25:
You haven't given me any information about what it DOES do.
At a quick look, I can see that you're using the increment and decrement operators incorrectly.
"seconds = seconds++;" is wrong. The line should be: "seconds++;"
http://en.cppreference.com/w/cpp/language/operator_incdec

closed 2016-11-25 by DeanIsMe

saurabhreigns, 2016-11-25:
Thanks Dean. I feel embarrassed to have committed such a silly mistake. Thanks for yor help. what the code is intended to do is use four buttons to set a timer and start it by hitting A4 button. the code is not working fully yet but I hope i will get it working soon. It would be great if you can point out to some other mistakes I may have done , but cant see, Thanks