untr0py/SevSeg #11: Digits 2 and 3 not working

petrovnn, 2016-12-12:
Hello, i have a problem: all digits except 2 and 3 is working fine. Instead  3 showing  breaking 0 (see the photo). Maybe this is a problem with library or this model of display not supported by library? Note: i'm using not original Arduino, but chinese copy of this from Aliexpress. Model is: E40361-I-BL-O-W 1043 9

your library mentioned here: http://www.instructables.com/id/Using-a-4-digit-7-segment-display-with-arduino/

[image]
[image]
[image]
[image]

DeanIsMe, 2016-12-12:
Hi,

Are you sure that you have the correct segment pins and digit pins?

Are you able to change digits 1 and 4?

Dean

[sam_0049]<https://cloud.githubusercontent.com/assets/1567284/21086166/ec2fc406-c02a-11e6-9726-6a4556df66de.JPG>
[sam_0050]<https://cloud.githubusercontent.com/assets/1567284/21086167/ec311f5e-c02a-11e6-9da5-806165e3255f.JPG>
[sam_0051]<https://cloud.githubusercontent.com/assets/1567284/21086169/ec3366e2-c02a-11e6-82d8-dea39327051d.JPG>
[sam_0052]<https://cloud.githubusercontent.com/assets/1567284/21086168/ec32e8ca-c02a-11e6-838a-d2edd1c1cabd.JPG>

-

petrovnn, 2016-12-12:
> Are you sure that you have the correct segment pins and digit pins?

i'm shure. I make this scheme again on different wireboard and cables, result same. Did pins exactly like in the tutorial:
http://www.instructables.com/id/Using-a-4-digit-7-segment-display-with-arduino/

**here we show "1234":**
[image]

if I change places gray wire to figures 1 and 4, will be the following:
[image]

maybe I can do some other test?

petrovnn, 2016-12-12:
i'm find some stuff in google by request: "e40361 datasheet"
http://www.toyo-led.com/wp-content/uploads/2013/12/E40361-LI.pdf

petrovnn, 2016-12-12:
so, digit "3" working with 
"B00011111"
instead yours
"B01001111"
in file https://github.com/DeanIsMe/SevSeg/blob/master/SevSeg.cpp

unfortunately, digit 2 still not working. Can't pick up right combination

DeanIsMe, 2016-12-13:
That tells me that you have it wired/configured incorrectly. You need to find out which pins are for which segments. Then, update  byte segmentPins[] = {6, 7, 8, 9, 10, 11, 12, 13}; accordingly.

The pins numbers in segmentPins[] need to be in the order A,B,C,D,E,F,G,DP, according to the following image.

[cid:dfc8438a-ef44-42cf-960a-f0e19e493a4d]

closed 2016-12-18 by DeanIsMe

qlereboursBS, 2017-12-28:
I had the same problem and i found a solution. I unplugged each pin to confirm that it was wired correctly.
I noticed that the F and G pins was reversed (my component was not the same as the one used for the tutorial: https://cdn.instructables.com/F1Z/ZJYV/IBTF513Q/F1ZZJYVIBTF513Q.MEDIUM.jpg). I wired it correctly and it worked.