untr0py/SevSeg #13: adding leading 0 to number

neroroxxx, 2017-01-16:
Hi, the library is great! my only issue is that i want to use a leading 0, i have a number that changes every so often and goes from 111 to 6488 when i'm on 111 i would like it to display 0111 to fill all digits, Is there an option to do this?

Thanks!

DeanIsMe, 2017-01-20:
Sorry, this is not an option.
If you want leading zeros, it's quite easy to edit the file SevSeg.cpp to achieve this.
All that you have to do, is comment out or delete the small for() loop underneath the comment:
"// Find unnnecessary leading zeros and set them to BLANK"

leres, 2017-02-12:
Here's a patch that makes leading zeros an option: [patch-SevSeg.cpp](https://gist.github.com/leres/d66ee74a0f3decd86ee08927152a8b9f)
To use it add define **ZERO_PAD**

DeanIsMe, 2017-03-29:
Implemented in version 3.3.0

closed 2017-03-29 by DeanIsMe