untr0py/SevSeg #21: Cannot display "."

Bowserw2c, 2017-05-04:
you have taken the input from pin 13 to dp pin of 7 segment display but there is no control in your code to turn the decimal point on.

DeanIsMe, 2017-05-05:
The decimal point will turn on when displaying a number, via the setNumber command. If I supported the decimal point via setChars command, it would take up an entire character (resulting in less space, and bad number rendering).
Example:
`setChars("3.14");      Will display [3 .14]` (Bad rendering)
Whereas
`setNumber(314, 2); Will display [ 3.14]` (Looks good)
That is why I chose not to include the decimal place in setChars().

closed 2017-05-05 by DeanIsMe

Bowserw2c, 2017-05-05:
Ok thanks for the explanation ... and thanks for the library!