untr0py/SevSeg #24: Problem with millis() overflow in SevSeg_Counter.ino example

rsocol, 2017-06-18:
The SevSeg_Counter.ino example seems not able to handle the millis() overflow (at about 49.7 days).
See https://arduino.stackexchange.com/questions/12587/how-can-i-handle-the-millis-rollover

DeanIsMe, 2017-06-18:
That's correct. So, if you're intending on using the example program for periods longer than 49.7 days (requiring a 9 digit display), then the time comparing should first be altered.

closed 2017-06-18 by DeanIsMe

rsocol, 2017-06-18:
As I understand, it's a best practice to compare durations instead of timestamps. That would require changing just two lines:
```
  if (millis()-timer >= 100) {
    deciSeconds++; // 100 milliSeconds is equal to 1 deciSecond
    timer = millis(); 
```
The problem is not about displaying 9 digits, because the example (as is, using 4 digits) would get stuck on overflow. Try a ```setMillis(-3000);``` in the Setup function to see that. Of course, I don't expect anyone to run such an example for 50 days, but it's a matter of principle: adopting the best coding practices in the examples.

DeanIsMe, 2017-06-19:
Thanks for the suggestion. I've made a change to the timer to implement good millis() practices.