untr0py/SevSeg #29: Leave decimal DOT unused

Chinoforest, 2017-12-10:
Hi. I'm developing an application where I will show only characters on the display. So, the decimal dot is not required and I would love to leave that pin free for another use. 

Reviewing your library (great work by the way!) I saw that it always asumes that the dot point is connected. It would be great if you could pass a flag on the begin function to tell the library is the dot has a pin assigned or not. So that way I can pass 7 pins instead of 8.

Thanks!!
Gonzalo

Chinoforest, 2017-12-10:
I've have modified the library. Here's a copy of it. I've just added another parameter to BEGIN that is a boolean that tells the function if the decimal dot is connected or not (dotPointNotConnected). Then I added a byte variable numSegments that retains that configuration (7 or 8 segments). Then everywhere where you were comparing agains a fixed 8, now compares agains numSegments. Just a small modification but now your library allows me to save an extra pin if I'm not planning on showing decimals.

Thanks!!
Gonzalo

[SevSeg.zip](https://github.com/DeanIsMe/SevSeg/files/1545994/SevSeg.zip)

Chinoforest, 2017-12-10:
And this is the updated example. I know that I should doing this via GitHub but I don't really know how to use it (pull request and all that stuff). I will learn someday, I promise haha.
Also, the benefit of leaving the dot point disconnected (if you are not using it) is reducing the flickering on the display (one less segment to multiplex).

[SevSeg_Counter.zip](https://github.com/DeanIsMe/SevSeg/files/1546008/SevSeg_Counter.zip)

DeanIsMe, 2018-08-11:
I quite like this solution. There would be a much nicer solution if Arduino did some sanity checking on the pin numbers - then you could just set the DP pin to an invalid value.
I have a new laptop and need to install GitHub and Arduino on it to make this commit properly. I'll try get around to it soon.

DeanIsMe, 2018-11-10:
Finally got around to this.

closed 2018-11-10 by DeanIsMe