untr0py/SevSeg #35: Brightness control

PCov3r, 2018-04-26:
Hello,
Could you please help me ? I'm trying to decrease the brightness of my seven segment display using your library.
The problem is that the brightness control doesn't work unless I put updateWithDelays on true.
And even when it seems to work, when I press a button, it just go back to 100% of brightness.
I'm indeed doing a clock and I need to set time and so on. Is that a problem with my code ?

Sorry for my bad  english, I'm french.

Keep up the good work.

`#include <SevSeg.h>
#include <DS3231.h>
#include <BY8001.h>
#include <Wire.h>
#include <SoftwareSerial.h>

SevSeg sevseg; //Instantiate a seven segment object
DS3231 clock;
RTCDateTime dt;
SoftwareSerial mp3Serial(19, 18);
BY8001 mp3;
const int HButton = 2;
const int MButton = 4;
const int SetTButton = 3;
const int SetAButton = 5;
const int ActivateA = 6;

int CurrentHour = 0;
int CurrentMin = 0;
int lastButtonTState = 0;
int ButtonTState = 0;
int ButtonHState = 0;
int lastButtonHState = 0;
int ButtonMState = 0;
int lastButtonMState = 0;

int lastButtonAState = 0;
int ButtonAState = 0; 
int HeureChoisie = 0;
int MinChoisie = 0;

int buttonStateActivateA = 0;
int alarmState = 0;
unsigned long previousMillis = 0;
byte CurrentStatus = 0;
int song_number = 0;

const int AnodeColon = 8;
const int CathodeColon = 52;

const int potPin = A2;    // select the input pin for the potentiometer
int val = 0; // variable to store the value coming from the sensor
int previousval = 0;
int interval = 1000;

void setup() {
   Wire.begin();
    // Music
  Serial.begin(9600);  // set serial monitor baud rate to Arduino IDE
  mp3Serial.begin(9600);  // BY8001 set to 9600 baud (required)
  mp3.setup(mp3Serial); // tell BY8001 library which serial port to use.
  delay(800);  // allow time for BY8001 cold boot; may adjust depending on flash storage size
  
   pinMode(HButton, INPUT);
   pinMode(MButton, INPUT); 
   pinMode(SetTButton, INPUT); 
   pinMode(SetAButton, INPUT); 
   pinMode(ActivateA, INPUT); 
   
   byte numDigits = 4;
[... 264 lines cut ...]
    AfficherH_A();
     } 
    }
  }
 }
}

 //Fonction définir volume
void Volume() {
  unsigned long currentMillis = millis();

if (currentMillis - previousMillis >= interval) {
    previousMillis = currentMillis;
sevseg.refreshDisplay();
val = analogRead(potPin);

val = map(val, 0, 1021, 0, 31);
val = constrain(val, 0, 30);
if(val != previousval){
sevseg.refreshDisplay();
mp3.setVolume(val);
previousval = val;
 }
}
}`

DeanIsMe, 2018-04-27:
Hi,

First, you can set the brightness to 0 instead of 1.
To decrease the brightness further, you need to call sevseg.refreshDisplay() less often.
I'm guessing that the reason the brightness changes when pressing a button, is due to changes in how often sevseg.refreshDisplay() is called. For example, in AddicherH() there is a delay(0.5) in the loop, which means that refreshDisplay() will be called less often and the display will be dimmer.

Also, there seems to be a lot of unnecessary calls of sevseg.refreshDisplay() in this program. These probably aren't doing any harm.

Regards,

Dean

PCov3r, 2018-05-02:
That works !
I simply put delay(1) in loops and now the brightness doesn't increase whenever I push a button.

Thanks a lot for your help. I can now put my alarm clock in a box.
Thank you very much !

closed 2018-05-13 by PCov3r