untr0py/SevSeg #36: Displaying 2 and 3

dk2dark, 2018-04-29:
I am having issues using th libary as when it has to display 2 and 3 it dos not light up the middle LED instead it lights up a diffrent one, not a random one the same every time it is the top left LED.

All other numbers work and i have also tried restarting and even tried 3 different 4 digit seven segment displays.

The code i am triing to run to get this to work:
#include "SevSeg.h"

SevSeg sevseg; //Initiate a seven segment controller object

void setup() {

byte numDigits = 4;

byte digitPins[] = {48, 52, 38, 50};

byte segmentPins[] = {26, 22, 34, 30, 28, 36, 24, 32};

sevseg.begin(COMMON_CATHODE, numDigits, digitPins, segmentPins);

sevseg.setBrightness(90);

}

void loop() {

sevseg.setNumber(3232, -1);

sevseg.refreshDisplay(); // Must run repeatedly

}

DeanIsMe, 2018-04-29:
I think you'll find that you have segment G (middle) mixed up with segment F.
Try this:
byte segmentPins[] = {26, 22, 34, 30, 28, 24, 36, 32};

closed 2018-04-29 by DeanIsMe

dk2dark, 2018-04-29:
Wauv now  just feel a little like a fool for having such a simple mistake cause yeah that works.
Out of curiosity how come this not affected other number using G? (like 6 8 and 9)

DeanIsMe, 2018-04-29:
On all numbers other than 2 and 3, the segments F and G are the same (either both on, or both off). So, they all end up being displayed correctly despite the mix up.