untr0py/SevSeg #37: 4 Digit 7 Segment display upper led always on

ItsVince1234, 2018-05-09:
Like the title says, here is my code:

// Arduino Light sensor code created by AR Builder.
#include "SevSeg.h"
SevSeg sevseg; 
void setup() {
 int A = analogRead(A0);
  Serial.begin(9600); // Start Serial.
  pinMode(13, OUTPUT);
  byte numDigits = 4;
  byte digitPins[] = {9, 10, 11, 12};
  byte segmentPins[] = {1, 2, 3, 4, 5, 6, 7, 8};

  bool resistorsOnSegments = false; 
  bool updateWithDelaysIn = true;
  byte hardwareConfig = COMMON_CATHODE; 
  sevseg.begin(hardwareConfig, numDigits, digitPins, segmentPins, resistorsOnSegments);
  sevseg.setBrightness(90);
}
  

void loop() {
 sevseg.blank();
int A = analogRead(A0);
{
   if (A < 150) {
     sevseg.setNumber(150,3);
    sevseg.refreshDisplay(); 
   }
   else{
     sevseg.blank();
   }
}
}

DeanIsMe, 2018-05-10:
Hi,

It's most likely that you don't have the right pins connected for your particular display. Confirm that each segment and digit pin is correct.

-Dean

emilianos, 2018-07-13:
Hi,
I'm experiencing the same issue:
Only the top segment of the first digit is always on.
I checked several times and it's properly connected.. also I think if it was a problem with that segment A, it should fail for every digit not just first one..

[image]

emilianos, 2018-07-13:
Hi I was having the same issue, and I found out that it happens when CP is not assigned a PIN:
```
byte segmentPins[] = {2, 3, 4, 5, 6, 7, 8, 13}; //Segments: A,B,C,D,E,F,G,CP
```
to avoid the issue assign any pin to CP (13 in this case), otherwise it will show a segment on on the first digit

DeanIsMe, 2018-08-11:
Hi @emilianos, note that @ItsVince1234  has pin numbers for all 8 segments defined

DeanIsMe, 2018-11-10:
I realised the problem tonight - it occurred because sevseg.refreshDisplay(); was no longer being called. I just make a commit that should improve the blank() functionality.
0116b56

closed 2018-11-10 by DeanIsMe