untr0py/SevSeg #42: Cant control brightness.

gavron04, 2018-09-04:
Hello. I have atmega32 on board with https://github.com/MCUdude/MightyCore library. All works ok with sevseg, but I cant control brightnes.. I have tested range 0-100 but displays is always on with same brightness.

Some ideas?

Regards

DeanIsMe, 2018-09-04:
Brightness control gives different effects on different hardware. Try enabling updateWithDelaysIn in the .begin() statement and see if your results change.
Do the 2 digits work normally in all other cases? What happens when you call blank()?

zhekaus, 2018-10-04:
The same issue. I've got 3641AH.

gavron04, 2018-10-09:
Some update?

How I can turn off display? Without hardware changes?

Regards

DeanIsMe, 2018-11-10:
Calling 'blank' sets the display to blank. If you call setNumber() or setChars() after blank(), then it will continue displaying. Is that what's happening?

Also, I just committed an update with changes to the brightness system. You now have a much larger range to try. Please give it a go and let me know how it goes (I unfortunately don't have the hardware to test this anymore).

gavron04, 2018-11-15:
"I just committed an update with changes to the brightness system." - but where I can find new files?

Regards,
Mateusz

gavron04, 2018-11-15:
Ok. I will test on Arduino uno (with atmega 328p) and also with Atmega32 ( https://github.com/MCUdude/MightyCore ). On Atmega32 I can test at the weekend, for atmega328p I need more free time to build prototype PCB.

0xdec, 2018-11-15:
@DeanIsMe the last commit is from Aug 21, are you sure you pushed your commit?

DeanIsMe, 2018-11-19:
Sorry for the confusion - it turns out that I didn't leave GitHub Desktop open long enough for it to sync the commits. They should all be visible now.

DeanIsMe, 2019-02-10:
Has anyone been able to test the new brightness system? I'd like to publish the latest state as a release.
@gavron04

gavron04, 2019-02-14:
Hi, I will test at the weekend. After test I will reply how it works.

Regards

DeanIsMe, 2019-03-29:
I'm going to assume that it works and close this issue.

closed 2019-03-29 by DeanIsMe