untr0py/SevSeg #44: 4-digit 7-segment display with colon

abuhafss, 2018-11-24:
Hi

How can this library be used with 4-digit modules having colon?

a) In some modules, both dots of the colon are used separately with D2 and D3. 
b) In some modules, both dots of the colon are used separately with D3 and D4.

DeanIsMe, 2018-12-01:
Hi,
So the upper dot of the colon a separately addressable segment? Which means there would be 9 segments for those digits.
Adding a segment would require quite a large number of changes in the library because BYTE (8-bit) variables are used to store which segments are being illuminated. These data types would need to be changed to SHORT (16-bits) to store information for 9 segments. There would also need to be a change in setChars() for setting it.

Do you need separate control of the top dot in the colon? Depending on what you need to display, you may be able to wire up both dots to illuminate together - which means you wouldn't need a separate control line,

swaroopbu, 2020-06-17:
how to control individual dots

DeanIsMe, 2020-06-17:
You can control individual dots with the setSegments command. From SevSeg.cpp:

```
// setSegments
/******************************************************************************/
// Sets the 'digitCodes' that are required to display the desired segments.
// Using this function, one can display any arbitrary set of segments (like
// letters, symbols or animated cursors). See setDigitCodes() for common
// numeric examples.
//
// Bit-segment mapping:  0bHGFEDCBA
//      Visual mapping:
//                        AAAA          0000
//                       F    B        5    1
//                       F    B        5    1
//                        GGGG          6666
//                       E    C        4    2
//                       E    C        4    2        (Segment H is often called
//                        DDDD  H       3333  7      DP, for Decimal Point)
void SevSeg::setSegments(byte segs[]) {
```

closed 2020-06-17 by DeanIsMe