untr0py/SevSeg #46: Apparent error

pedroypatana, 2018-12-28:
I'm using the library. With one digit everything is fine when I use two digits and I want to draw the number 11 in the display 71 ???
------------------------
#include "SevSeg.h"
SevSeg sevseg; //Initiate a seven segment controller object
void setup() {
  pinMode (A0,OUTPUT);
  pinMode (A1,OUTPUT);
    byte numDigits = 2; 
  // byte digitPins[] = {5,6,7,8};
    byte digitPins[] = {5,6};
    byte segmentPins[] = {9,10,11,12,13,A0,A1};
    sevseg.begin(COMMON_CATHODE, numDigits, digitPins, segmentPins);
    sevseg.setBrightness(100);
    sevseg.setNumber(11);
}
void loop() {
       sevseg.refreshDisplay(); // Must run repeatedly
}
----------------------------

otro codigo
----------------
#include "SevSeg.h"

SevSeg sevseg; //Initiate a seven segment controller object

void setup() {

  pinMode (A0,OUTPUT);
  pinMode (A1,OUTPUT);
    byte numDigits = 2; 
  // byte digitPins[] = {5,6,7,8};
    byte digitPins[] = {5,6};
    byte segmentPins[] = {9,10,11,12,13,A0,A1};
    sevseg.begin(COMMON_CATHODE, numDigits, digitPins, segmentPins);
    sevseg.setBrightness(100);
    sevseg.setNumber(31);
}

void loop() {
   
    sevseg.refreshDisplay(); // Must run repeatedly
}
"Here he draws 31 without problems. I appreciate some help!!!!!
[image]
[image]

"

DeanIsMe, 2018-12-29:
It's hard to diagnose the problem without actually being there. You can try to display "11" and swap the wires into D5 and D6 - if the problem stays exactly where it is, then the problem is entirely hardware. If the problem changes to the right-digit, then the problem is software (in which case I may need to fix the library).
My guess is that segment A is turning on whenever B is on (or perhaps A  is turning on whenever C is on). Can you try to display all numbers 0-9 and let me know what it actually displays?

[img_20181228_203527805]<https://user-images.githubusercontent.com/46231207/50530854-a34a0500-0ae0-11e9-85c2-691c26b0722c.jpg>
[img_20181228_203443677]<https://user-images.githubusercontent.com/46231207/50530855-a34a0500-0ae0-11e9-9fcf-9148c8983dfe.jpg>

"

pedroypatana, 2018-12-30:
Hello Goodnight. I think he solved the problem. If you look at the sevenseg array I'm using 7 values instead of 8
I did this because I do not use the decimal point. But even if I do not use that led; I think it should go in that array
Changing byte segmentPins [] = {9,10,11,12,13, A0, A1, A2} now seems to work.
Anyway, thanks for answering
Pedro

DeanIsMe, 2018-12-30:
Good to see!
If you download the latest version from GitHub, there's a field added 'disableDecPoint' which allows you to not use a pin for the decimal.

closed 2018-12-30 by DeanIsMe