untr0py/SevSeg #49: Freeze with I2C Call

maymayday90, 2019-02-06:
My Teensy LC is freezing when I include a call to I2C. Remove that call, works fine.

Found one other person with this issue, they fixed themselves. I am not experienced enough to figure out what they did...

https://forum.arduino.cc/index.php?topic=497128.0

My code:

`#include <SevSeg.h>
#include <Adafruit_Sensor.h>
#include <Adafruit_BMP280.h>

  Adafruit_BMP280 bme; // I2C
  SevSeg sevseg; 

  int buttonState = 1; 
  const int buttonPin = 11;
  const int buzzer = 23;
  int currentALT = 1;

void setup(){
  byte numDigits = 4;
  byte digitPins[] = {17, 16, 15, 14};
  byte segmentPins[] = {0,1,2,3,4,5,6,7};

  bool resistorsOnSegments = true; 
  bool updateWithDelaysIn = true;
  byte hardwareConfig = COMMON_ANODE; 
  sevseg.begin(hardwareConfig, numDigits, digitPins, segmentPins, resistorsOnSegments);
  sevseg.setBrightness(90);

  pinMode(buzzer, OUTPUT);
  pinMode(buttonPin, INPUT);
}

void loop(){

    buttonState = digitalRead(buttonPin);
    currentALT = bme.readAltitude(1013.25); //including this causes the Teensy not to operate (nothing on 7dig display)
    delay(10);
    

  if (buttonState == HIGH) {

    sevseg.setNumber(currentALT, 0);
    sevseg.refreshDisplay();  
    tone(buzzer, 2000);
  
  } else {

    sevseg.setNumber(1234, 0);
    sevseg.refreshDisplay(); 
    noTone(buzzer); 
  }
}`

DeanIsMe, 2019-02-06:
Hi,

Sorry, I'm not familiar with the Adafruit_BMP280 library and I don't have time to investigate further. I see that in the forum linked<https://forum.arduino.cc/index.php?topic=497128.0>, the poster fixed the problem by writing their own I2C code. Perhaps you could try a different I2C library?

Also, you should remove the delay(10) from your loop.

Regards,

Dean

https://forum.arduino.cc/index.php?topic=497128.0

maymayday90, 2019-02-06:
OK, thanks for the reply Dean. Will take a look at alternate methods for interfacing with the BMP280.

Had the delay for trying to troubleshoot, will remove.

Thanks!

maymayday90, 2019-02-13:
Tried different libraries, also tried an Uno instead of the Teensy, still freezing as soon as I try to get info from the barometer.......

```
#include <SevSeg.h>
#include "Seeed_BMP280.h"
#include <Wire.h>

  BMP280 bmp280;
  SevSeg sevseg; 

  const int buttonPin = 13;
  const int buzzer = 9;
  int buttonState = 1; 
  int currentALT = 1;
  float pressure;

void setup(){
  //Serial.begin(9600);
  byte numDigits = 4;
  byte digitPins[] = {8, 10, 11, 12};
  byte segmentPins[] = {0,1,2,3,4,5,6,7};

  bool resistorsOnSegments = true; 
  bool updateWithDelaysIn = true;
  byte hardwareConfig = COMMON_ANODE; 
  sevseg.begin(hardwareConfig, numDigits, digitPins, segmentPins, resistorsOnSegments);
  sevseg.setBrightness(90);

  pinMode(buzzer, OUTPUT);
  pinMode(buttonPin, INPUT);
}

void loop(){

    buttonState = digitalRead(buttonPin);
    //pressure = bmp280.getPressure();
    //currentALT = bmp280.calcAltitude(pressure);

  if (buttonState == HIGH) {

    sevseg.setNumber(currentALT, 0);
    sevseg.refreshDisplay();  
    tone(buzzer, 2000);
    pressure = bmp280.getPressure();
  
  } else {

    sevseg.setNumber(1234, 0);
    sevseg.refreshDisplay(); 
    noTone(buzzer); 
  }
}
```

maymayday90, 2019-02-13:
Tried a different address (default is 0x77, switched to 0x76) and it works! That was several days of frustration. :P 

Thanks for the library!

closed 2019-02-13 by maymayday90