untr0py/SevSeg #50: dynamic brightness conytol and blank display

Zerplin, 2019-02-24:
Hi, i have a setup where i use a photo resistor's values mapped from -200 to 200 to dynamically control the brightness of the display. I tested the feature by shining a light at the photo resistor from varying distances and angels. It seemed to work most of the time, but sometimes when varying the angle or distance quickly the display would just go blank until i changed the angle or distance within a threshold. 

Might still have something to do with my code, but when i monitored the value of the brightness the value was always within -200 and 200. 

I suspect it's might have something to do with sudden and large changes of brightness or screen updates syncing.

DeanIsMe, 2019-02-24:
Thanks for the report! I just made a commit which should have fixed the issue. Please check and confirm that it's resolved.
e34b9693416d4dc69e52a28f78a5b326637d7156

Side note: the expression map(analogRead(A2),0,1000,-200,200); can return a number greater than 200, as an analog read value of 1023 would map to 209. But, the library lowers the value back down to 200 so this isn't a problem.

Zerplin, 2019-02-24:
Thank you for the fix! I downloaded the update and everything seems to work now. 
This was my first time submitting an issue on github and i'm very glad you could use it :D

closed 2019-02-24 by Zerplin