untr0py/SevSeg #51: write to curtain digit on a display

machmar, 2019-03-14:
Is there any way I could do that? Like for example. I want to use one display for displaying three one digit hex values. Each one on a different position, all three of the same time. How do I do that without writing/using a different library?

DeanIsMe, 2019-03-15:
If I understand your problem, this should be fairly easy to achieve in SevSeg.
Say you have the 3 numbers you wish to display (values 0-F): `byte num1, num2, num3;`
The code to set the display would look something like this:
`short toDisplay = (short) num1 * 0x100 + (short) num2 * 0x10 + (short) num3;`
`sevseg.setNumber(toDisplay, -1 true);`
And don't forget to make `leadingZerosIn = true;` in the `setup()` function.

machmar, 2019-03-16:
thanks! that worked. And aslo, is there any easy way to modify the linrary so it can do custom bases or at least base 20 (I realised I need to set a number from 0 not 1 to 16 on one digit and that not good for hexa)

DeanIsMe, 2019-03-16:
This would also be quite simple, because findDigits() uses a generic method of finding the digits. Just create an array called `powersOf20` which has stores 20^x. (see `powersOf16`). Then, in `findDigits()`, you need to make `powersOfBase` point to this new array.

closed 2019-03-16 by DeanIsMe

machmar, 2019-03-16:
Ok thanks I'm gonna try it!
I'll let you know and maybe fork the library if I am able to make it fully work.

machmar, 2019-03-17:
It works!