untr0py/SevSeg #68: leading zeros?

L8zin, 2020-03-30:
Hello, I might've caught the stupid virus here but I can't figure out how to get leading zeros working. I have a float as the output with 1 decimal place:

`  sevseg.setNumber(average,1);`

This is my setup:

`byte numDigits = 4;
byte digitPins[] = {13, 12, 11, 10};
byte segmentPins[] = {2, 3, 4, 5, 6, 7, 8, 9};
bool leadingZeros = true;
sevseg.begin(COMMON_CATHODE, numDigits, digitPins, segmentPins, leadingZeros); `

L8zin, 2020-03-30:
I am very new to this and I created this account just to post this

L8zin, 2020-03-30:
I seem to have fixed my own problem by changing `if (!leadingzeros)` to `if (leading zeros = false)` in the SevSeg.cpp. You might want to look into this though.

DeanIsMe, 2020-03-30:
Hmmm `leadingZeros` is `bool`, so `!leadingZeros` should be equivalent to `leadingZeros == false`.
If anyone else is having an issue, please post here.

machmar, 2020-03-30:
I just wanna comment that `if (leading zeros = false)` means that it will always skip the if loop.

DeanIsMe, 2020-08-02:
Yes, and the loop removes the leading zeros. Hence, in the case of L8zin the leading zeros will remain and be displayed. 
Given the lack of activity, I will close this issue.

closed 2020-08-02 by DeanIsMe