untr0py/SevSeg #70: Show numbers and letters at the same time

machmar, 2020-05-08:
Is there any way I can display numbers and letters at the same time? For example: 2.5A (2.5 ampers). 
Is there any way or do I have to try to edit the library?

bridystone, 2020-05-08:
Hi @machmar,

think of it a little differently:
What you want is to print letters.
char[4] test = {'2', '.', '5', 'A'} // "2.5A"

so the following code should work:
```
    float number = 2.5;
    char buf[5]; //4 characters + 1 dot = 5 char
    sprintf("%3.1fA", buf, number); // %3.1f ==> 3 char alltogehter | 1 char after the dot
    SevSeg.setChars(buf);

```

machmar, 2020-05-08:
Thanks @bridystone ! I never heard of `sprintf()`. What does it do ?
Also, I have only 3 digits but that should be fine.

machmar, 2020-05-08:
Woah the `sprintf()` is pretty cool! Thanks so much - such a simple solution.

closed 2020-05-08 by machmar

machmar, 2020-05-08:
Too bad the `sprintf()` in Arduino doesn't work with floats

bridystone, 2020-05-08:
sorry about that - i didn't know.
a short google search brought up [this](https://www.hobbytronics.co.uk/arduino-float-vars) probable solution:
```
float number = 2.5;
char buf[6]; //4 characters + 1 dot + \0 for ending (i forgot before) = 6 char 
dtostrf(number, 4, 1, buf);
//now adding the 'A' character
char output[7];
sprintf("%sA", output, buf);
SevSeg.setChars(buf);

```
This should work on Arduino.

i'm not 100% sure about the '\0' at the end. usually this character ends an string in C.

machmar, 2020-05-08:
This is how I did it:
`float number = 2.5;
  char buf[4]; //3 characters + 1 dot = 4 char
  sprintf(buf, "%d.%dA", (int) number, (int)(number * 10.0) % 10); // I split the float and than write it as two integers with a dot between them
  sevseg.setChars(buf);`

machmar, 2020-05-08:
Thanks for the help !!