untr0py/SevSeg #71: First segment stays lit always if only seven segmentPins are specified (DecPoint is set as disabled)

seanspotatobusiness, 2020-05-11:
Hi. I am using SevSeg.h in an Arduino sketch on an Arduino Pro Mini 5V 16 MHz 328. I have a three-digit seven-segment display and I don't wish to use the decimal point. I have the following code in the setup section of my sketch:

`byte numDigits = 3;
byte digitPins[] = {2, 3, 4};
byte segmentPins[] = {5, 6, 7, 8, 9, 10, 11};
bool disableDecPoint = true;`

Since the decimal point is set to disabled, I think I shouldn't have to specify a pin to use but if I end the segment pins list at 11, I get some sort of glitch where in the first segment of the display (segment A for the first digit) is always illuminated. If I add 12 to the end of the list, the problem goes away but then pin 12 is kind of wasted.

The entire code of the sketch is below and a picture of the lit segment is below that. It makes a one look like a seven and a four look like a nine and is also lit when there is not meant to be a digit displayed in that location at all.

`#include "SevSeg.h"

SevSeg sevseg; //Initiate a seven segment controller object

byte potentiometerPin = 14;

int timerSetting = 0;

void setup() {

byte numDigits = 3;
byte digitPins[] = {2, 3, 4};
byte segmentPins[] = {5, 6, 7, 8, 9, 10, 11};
bool disableDecPoint = true;

pinMode(13, OUTPUT);
digitalWrite(13, LOW);
pinMode(15, OUTPUT);
digitalWrite(15, HIGH);

sevseg.begin(COMMON_CATHODE, numDigits, digitPins, segmentPins);

sevseg.setBrightness(100);

}

void loop() {

timerSetting = map(analogRead(potentiometerPin), 0, 1023, 10, 200);

sevseg.setNumber(timerSetting);
sevseg.refreshDisplay(); // Must run repeatedly

}`

[image]

PBudmark, 2020-07-13:
Hi, 
in your call to sevseg.begin() you are missing some required variables to be able to specify disableDecPoint=true,
therefore the default setting disableDecPoint=false is used.

The result is that 8 pin numbers are read out of your variable segmentPins, and the pin number used for decimalPoint can by any garbage (probably zero)

You need to include the missing values to be able to specify disableDecPoint=false, as all information must be passed to begin():

```
  sevseg.begin(COMMON_CATHODE, numDigits, digitPins, segmentPins, 
    false, false, false, disableDecPoint); 
```
or even:

```
   bool resistorsOnSegments = false; // 'false' means resistors are on digit pins
   bool updateWithDelays = false; // Default 'false' is Recommended
   bool leadingZeros = false; // Use 'true' if you'd like to keep the leading zeros
   bool disableDecPoint = true; // Use 'true' if your decimal point doesn't exist or isn't connected

   sevseg.begin(COMMON_CATHODE, numDigits, digitPins, segmentPins, 
     resistorsOnSegments, updateWithDelays, leadingZeros, disableDecPoint);
```

seanspotatobusiness, 2020-07-31:
Thanks very much!

closed 2020-07-31 by seanspotatobusiness