untr0py/SevSeg #78: Live simulations for the library 😀

Puneethkumarsn, 2020-11-04:
Hi There

I am from Wokwi team. We have created a sample simulation fo the library here
https://wokwi.com/arduino/libraries/SevSeg/SevSeg_Counter

We will be glad to create more and share the links to you so that you can put it in your readMe files for your library users. It will give instant confidence and also teh feel of what the library will do.

We have many such libraries as examples and the links I will share are permanent links. I hope you will find this interesting. If you like this, Kindly let us know. We can create a PR for the ReadMe file with the simulation links fo your library code examples.

thank you
Puneeth

DeanIsMe, 2020-11-05:
Hi Puneeth,
Thanks for putting this together! I added a link to the Wokwi in the readme with fdb74aea438b82bf21f92d21eee5910a23ecc896.
-Dean

closed 2020-11-05 by DeanIsMe

Puneethkumarsn, 2020-11-08:
Dear Dean,

thank you. 
I would like to share the links for the other two examples here we have built. these are permanent links and you may add it as well.
Many thanks

Here are the links:

https://wokwi.com/arduino/libraries/SevSeg/stringWithPeriod
https://wokwi.com/arduino/libraries/SevSeg/testWholeDisplay

Also, we will be glad If you can share any suggestions and feedback on our open -source-based Arduino simulation tool 😃😃here:
https://wokwi.com

Many Thanks in Advance

DeanIsMe, 2020-11-08:
Thanks! I added these links.
I think that some cool features for Wokwi would be:  
- Control the rate of time movement
- Ability to 'probe' variables or signals to produce a graph of their value over time