untr0py/SevSeg #80: Blank Digit becomes to Zero if DP is on

kolaha99, 2021-01-26:
Example testWholeDisplay.ino
If your sketch fixed and DP goes from right digit to left then
if change 8888 to 1 in digits 0-1-2 when DP lid you will see "0" instead to be blank (leadingZeros = false)

dirtbikr, 2021-02-12:
@kolaha99 I have this issue as well and will be looking into it soon since I need it "fixed" (I put that in quotes as I've only lightly used this library and don't know any intricacies as of yet). It may be proper behavior and we are missing something obvious?

`
  byte numDigits = 4;   
  byte digitPins[] = {21, 26, 25, 27}; //Digits: 1,2,3,4 <--put one resistor (ex: 220 Ohms, or 330 Ohms, etc, on each digit pin)
  byte segmentPins[] = {19, 33, 12, 16, 17, 18, 14, 13}; //Segments: A,B,C,D,E,F,G,Period
  bool resistorsOnSegments = true; // 'false' means resistors are on digit pins
  byte hardwareConfig = COMMON_CATHODE; // See README.md for options
  bool updateWithDelays = false; // Default 'false' is Recommended
  bool leadingZeros = false; // Use 'true' if you'd like to keep the leading zeros
  bool disableDecPoint = false; // Use 'true' if your decimal point doesn't exist or isn't connected
`

With the above parameters and calling `sevseg.setNumber(f, 1);` the display works as expected. Calling `sevseg.setNumber(f);` or `sevseg.setNumber(f, -1);` results in the leading 0's setting apparently being ignored. 

I will update this issue when I find more information!

DeanIsMe, 2021-02-12:
Thanks for posting your setup, @dirtbikr . I can't identify an issue in the code on a first check.
To help me find the issue, can you also provide:

- the value & type of of `f` when you call `sevseg.setNumber(f, -1);`,
- what is displayed
- what you expect to be displayed

dirtbikr, 2021-02-13:
Hey @DeanIsMe , thanks for the reply!

f is an integer between 65-95

When f is 65, 0065 is displayed.

I expect blank blank 65 to be displayed.

DeanIsMe, 2021-02-13:
Ah, I think I found the issue. I'm guessing you're using a non-standard arduino platform, such as Teensyduino?
`decPlaces` uses `char` data type.
On arduino, 'char' data types are 'int8_t'. On Teensy, 'char' data types are 'uint8_t'.
So, on Teensy, `decPlaces = -1` gets interpreted as `decPlaces = 255`.

This is a relatively simple fix of changing the data type of decPlaces, but it opens to door to a bunch of data type compatibility issues. I'll update the library to get around these, but it will take a day or 2 to get the changes right.

dirtbikr, 2021-02-13:
That is correct, I am using an ESP32. 

Where do you find the definition for the char datatype? I would like to check it against my ESP32 board definitions.

Thanks for looking into this!

DeanIsMe, 2021-02-13:
I don't know where it's defined, but this should work to check if char is signed or unsigned:

```
int myVal = (int)((char)-1);
if (myVal == -1) {
  Serial.println("char is signed (arduino default)");
}
else {
  Serial.println("char is UNsigned");
}

```

DeanIsMe, 2021-02-13:
I was able to publish a new release (3.6.0) with the updated data types. Let me know if it fixes your issue.
Relevant commit: 5d8260a9f8351b094e5542e8e1db1a69ffce1d2e

dirtbikr, 2021-02-13:
Using the new version seems to have broken the `setChars()` method. 

One of my statements is `sevseg.setChars("err");`. Compilation fails with the message `invalid conversion from 'const char*' to 'const int8_t* {aka const signed char*}'`.

I did comment the `sevseg.setChars("err");`. line out and tested your fix and it appears to work as expected now! Great call.

DeanIsMe, 2021-02-14:
Oh damn I messed up setChars. Should be fixed now in v3.6.1!
Thanks for testing. I'll close this issue.

closed 2021-02-14 by DeanIsMe