untr0py/SevSeg #82: Issue with .setChars()

dcadarko, 2021-02-18:
When using .setChars, the seven segment displays don't work as they should after turning off the arduino after sketch upload and turning it on.

DeanIsMe, 2021-02-19:
Hi @dcadarko, can you list the steps required to produce the error and describe what the display does.
Cheers

dcadarko, 2021-02-19:
```
#include <dht.h>
#include <SevSeg.h>
SevSeg sevseg;
SevSeg sevseg2;
dht DHT;
#define DHT11_PIN A2

void setup() {
  byte numDigits = 1;
  byte digitPins[] = {};
  byte segmentPins[] = {11,12, A1, A0, 13, 10, 9 };
  byte hardwareConfig = COMMON_CATHODE;
  bool resistorsOnSegments = true;
  sevseg.begin(hardwareConfig, numDigits, digitPins, segmentPins, resistorsOnSegments);
  sevseg.setBrightness(90);
  byte numDigits2 = 1;
  byte digitPins2[] = {};
  byte segmentPins2[] = {6, 5, 3, 2, 4, 7, 8};
  bool resistorsOnSegments2 = true;
  byte hardwareConfig2 = COMMON_CATHODE; 
  sevseg2.begin(hardwareConfig2, numDigits2, digitPins2, segmentPins2, resistorsOnSegments2);
  sevseg2.setBrightness(90);
  Serial.begin(9600);
  pinMode(12, OUTPUT);
  pinMode(A1, OUTPUT);
  pinMode(11, OUTPUT);
  pinMode(6, OUTPUT);
  pinMode(13, OUTPUT);
  pinMode(10, OUTPUT);
  pinMode(9, OUTPUT);
}

void h(){
  digitalWrite(A1, HIGH);
  digitalWrite(12, HIGH);
  digitalWrite(13, HIGH);
  digitalWrite(10, HIGH);
  digitalWrite(9, HIGH);
}

void t(){
  digitalWrite(12,HIGH);
  digitalWrite(A1, HIGH);
  digitalWrite(11, HIGH);
  digitalWrite(6, HIGH);
}

void loop() {
  int chk = DHT.read11(DHT11_PIN);
  int temp = DHT.temperature;
  int hum = DHT.humidity;
  int firsthum = (hum/10) % 10;
  int secondhum = hum % 10;
  int first = (temp/10) % 10; 
  int second = temp % 10;
  delay(1000);
  sevseg.setNumber(first);
  sevseg2.setNumber(second);
  sevseg2.refreshDisplay();
  sevseg.refreshDisplay();
  Serial.println(temp);  
  delay(3000);
  sevseg.blank();
  h();
  sevseg2.blank();
  delay(3000);
  sevseg.setNumber(firsthum);
  sevseg2.setNumber(secondhum);
  sevseg2.refreshDisplay();
  sevseg.refreshDisplay();
  delay(3000);
  sevseg2.blank();
  sevseg.blank();
  t();
  delay(1000);
}
```
This is the code I've used that caused this issue, sorry it's kinda bad, but when I upload the code on the arduino, the code works ok, but after i unplug my arduino from my computer and plug it again on the computer or a power adapter, the seven segment displays start flickering.

DeanIsMe, 2021-03-04:
@dcadarko you're using delays, which don't work with the sevseg library (as it uses multiplexing). You're also calling `h()` and `t()` which would impact the display.
Please read the Software section of the documentation.

closed 2021-03-04 by DeanIsMe