untr0py/SevSeg #85: setChars to accept ascii codes?

256k, 2021-04-14:
Hi I'm sorry if this is the wrong place to write this but i've been burning my brain trying to figure this out and i can't find any resources to help me.

i'm trying to get sevseg to display the alphabetical characters using ascii code numbers but i can't figure it out.

it doesn't work when passing the number to setChars()   nor does it work when converting the number to a char by char myChar = ascii_number

this is my code here
`#include "SevSeg.h"
SevSeg sevseg; 
int count= 65;
char alphabet[] = "abcdefghijk";

void setup(){
  Serial.begin(9600);
    byte numDigits = 1;
    byte digitPins[] = {};
//    byte segmentPins[] = {6, 5, 2, 3, 4, 7, 8, 9}
 byte segmentPins[] = {4, 5, 6, 7, 8, 9, 14, 15};
    bool resistorsOnSegments = true;

    byte hardwareConfig = COMMON_CATHODE; 
    sevseg.begin(hardwareConfig, numDigits, digitPins, segmentPins, resistorsOnSegments);
    sevseg.setBrightness(90);
}

void loop(){
 
  delay(500);
  
 
    Serial.println(count);
    char myChar = count;
    Serial.println(myChar);
    sevseg.setChars(myChar);
//    sevseg.setChars(count);
             count = count + 1;

  
        sevseg.refreshDisplay();   
}`

hopefully im just dumb and there's a simple way otherwise i feel like this would be a beneficial feature.

thank you

DeanIsMe, 2021-04-14:
Does that even compile? You're sending an integer to `setChars` which requires a character array input.
You're also using delay(), which won't work with this library. Please read the documentation.
Displaying the character related to an ascii number is easy.
`char myCharArr[] = "A";`
`sevseg.setChars(myCharArr);`
To increment to the next number:
`myCharArr[0]++;`
To set to an ASCII number:
`myCharArr[0] = 65;`

closed 2021-04-14 by DeanIsMe

256k, 2021-04-14:
Thank you! i'm sorry for the stupid question i really appreciate your help. the string/character/array system in C is still quite confusing to me.