untr0py/SevSeg #89: Problem in begining the display

JaimeBa27, 2021-06-23:
Whatever I try I get this error all the time. What should I do? I also try with the examples.
Here is the error:

> Arduino:1.8.15 (Windows 7), Tarjeta:"Arduino Uno"
> C:\Users\ABEJA\Desktop\Arduino\LedDisplay\LedDisplay.ino: In function 'void setup()':
> 
> LedDisplay:14:9: error: 'class SevSeg' has no member named 'begin'; did you mean 'Begin'?
> 
>   sevseg.begin(hardwareConfig, numDigits, digitPins, segmentPins, resistorsOnSegments,
> 
>          ^~~~~
> 
>          Begin
> 
> exit status 1
> 
> 'class SevSeg' has no member named 'begin'; did you mean 'Begin'?

And here is the code:
#include <SevSeg.h>
SevSeg sevseg;

void setup() {
   byte numDigits = 4;
  byte digitPins[] = {2, 3, 4, 5};
  byte segmentPins[] = {6, 7, 8, 9, 10, 11, 12, 13};
  bool resistorsOnSegments = false; // 'false' means resistors are on digit pins
  byte hardwareConfig = COMMON_ANODE; // See README.md for options
  bool updateWithDelays = false; // Default 'false' is Recommended
  bool leadingZeros = false; // Use 'true' if you'd like to keep the leading zeros
  bool disableDecPoint = false; // Use 'true' if your decimal point doesn't exist or isn't connected. Then, you only need to specify 7 segmentPins[]

 sevseg.begin(hardwareConfig, numDigits, digitPins, segmentPins, resistorsOnSegments,
  updateWithDelays, leadingZeros, disableDecPoint);

}

void loop() {
}

DeanIsMe, 2021-06-23:
That's odd... How did you install the SevSeg library? Did you use the arduino Library Manager?

You could try downloading SevSeg.cpp and SevSeg.h from GitHub and copying them to the folder C:\Users\ABEJA\Desktop\Arduino\LedDisplay\

DeanIsMe, 2021-07-15:
Closing due to inactivity.

closed 2021-07-15 by DeanIsMe