untr0py/SevSeg #91: redeclaration of 'uint8_t segs [3]

Dein-Freund, 2021-08-18:
Hey, 
so I'm fairly new to this whole arduino thing and currently  I'm trying to run a seven-segment-display. But the program refuses to start and I don't get why. Maybe someone knows a solution.
I have attached a screenshot of the program.

Dein-Freund, 2021-08-18:
[image]

DeanIsMe, 2021-08-19:
Hi your-friend, the error tells you that `segs` has been declared twice, which is not allowed in C++. Change the highlighted line to segs[2] = 0x6D.

closed 2021-08-19 by DeanIsMe

Dein-Freund, 2021-08-19:
Maybe I got you wrong, but this doesn't work either.

[image]

machmar, 2021-08-19:
You are **redeclaring** (declaring multiple times) the variable `segs`.  
By writing `uint8_t segs[3] = {0, 0, 0};` you **declare** (create) the array `segs` and you **initialize** it (give it values), in this case, that's all zeroes. When you write `uint8_t segs[3] = {0, 0, 0};` again, you are **declaring** it again, even tho it has already been **declared**. Thus, a **redeclaration** happens.  What you really want to do, is to just give the array new values. Which is done by writing `segs = {0, 0, 0};`. By removing the **datatype** (`uint8_t`) you are only changing the values of the already existing (**declared**) variable. Also, there is no need for the number (`[3]`) because you are writing to all elements of the array at once.  
  
tl;dr:  
change the second`uint8_t segs[3] = {0, 0, 0x6D};` to `segs = {0, 0, 0x6D};`.  
Optionally, you could write `segs[2] = 0x6D;`. The `[2]` means that we are writing to the element number two. And there is no need for braces (`{}`) as we are writing a single value.  
  
Hope this helps! ☺