untr0py/SevSeg #95: sevensegment dipaly with serial

microElabDevelop, 2022-02-07:
hi. i try to run simple counter and value receive through Serial .but my disply blink when serial data come any solution here is my code.
`void setup(){
  Serial.begin(9600);
  pinMode(b1,INPUT);

//////////////////////
  bool resistorsOnSegments = false; // 'false' means resistors are on digit pins
  byte hardwareConfig =N_TRANSISTORS; // See README.md for options
  bool updateWithDelays = false; // Default 'false' is Recommended
  bool leadingZeros = true; // Use 'true' if you'd like to keep the leading zeros
  bool disableDecPoint = false; // Use 'true' if your decimal point doesn't exist or isn't connected. Then, you only need to specify 7 segmentPins[]

  sevseg.begin(hardwareConfig, numDigits, digitPins, segmentPins, resistorsOnSegments,
  updateWithDelays, leadingZeros, disableDecPoint);
  sevseg.setBrightness(100); //Note: 100 brightness simply corresponds to a delay of 2000us after lighting each segment. A brightness of 0 
//////////////////////////////////////////////////////////
Serial.println("Display");
}
///////////////////////////////////////////////////////////////////////////
void loop(){
unsigned long cM= micros();if(cM-pM>= interval) {pM=cM;sevseg.refreshDisplay();}

data1=getValue(incommingStr,',',0);
data2=getValue(incommingStr,',',1);
data1.toCharArray(charData1,data1.length()+1);
data2.toCharArray(charData2,data2.length()+1);
index=atoi(charData1);
limit=atoi(charData2);

//Serial.print("Index =");Serial.println(charData1);
//Serial.print("Limit =");Serial.println(charData2);
//Serial.println("//////////");
disSwitch=digitalRead(b1);///Serial.println(disSwitch);
if(disSwitch==LOW){segDis=index;}else{segDis=limit;}

/////////////
sevseg.setNumber(segDis,4);
}///End Main
///////////////////////////////
void serialEvent(){
while(Serial.available()){// wait till data to be filled in serial buffer
incommingStr =Serial.readStringUntil('\r\n'); // read the complete string
}}
String getValue(String data, char separator, int index){
  int found = 0;
  int strIndex[] = {0, -1};
  int maxIndex = data.length()-1;

  for(int i=0; i<=maxIndex && found<=index; i++){
    if(data.charAt(i)==separator || i==maxIndex){
        found++;
        strIndex[0] = strIndex[1]+1;
        strIndex[1] = (i == maxIndex) ? i+1 : i;
    }
  }

  return found>index ? data.substring(strIndex[0], strIndex[1]) : "";
}`

DeanIsMe, 2022-02-07:
What's the value of pM?

microElabDevelop, 2022-02-07:
i try 1 mill then i try 10 micro but same issue

DeanIsMe, 2022-02-08:
You must repeatedly run `sevseg.refreshDisplay();`
Your loop runs `Serial.readStringUntil('\r\n');`, which will block all code for 1 second. That means you are only refreshing the display once per second.

closed 2022-02-08 by DeanIsMe

microElabDevelop, 2022-02-10:
use this code i solve my string delay issue.
`if(Serial.available()){
char rxData=Serial.read();//Serial.print(rxData);
if(rxData=='\n'){strComplete=HIGH;rxData=0;i=0;}
else{strArry[i++]=rxData;}
}`

thankx

> You must repeatedly run `sevseg.refreshDisplay();` Your loop runs `Serial.readStringUntil('\r\n');`, which will block all code for 1 second. That means you are only refreshing the display once per second.

use this code i solve my string delay issue.
`if(Serial.available()){
char rxData=Serial.read();//Serial.print(rxData);
if(rxData=='\n'){strComplete=HIGH;rxData=0;i=0;}
else{strArry[i++]=rxData;}
}`

thankx