untr0py/SevSeg #96: Colon Support

RodjGaming, 2022-04-12:
Hi there,

I have a 4 digit 7-segment display with an additional colon and a 3rd upper dot(to the right of the 3rd digit). These dots(L1,L2,L3) have their own common cathode/anode with the other end connected to segments A, B, and C individually, basically making it its own digit. Below are the pictures from the datasheet of the display:
[image]
[image]

I would like if native colon support could be added for components like this, or if there is another way to achieve lighting up L1 and L2 currently in the library.

basgoossen, 2023-01-15:
Hi RodjGaming,
Is there a reason you're not just using digitalWrite for this, to control the respective pins?
If you want to control brightness of those dots you can also use analogWrite if the pins they are connected to support pwm.

machmar, 2023-01-15:
Hi, the issue is that these dots are multiplexed like if they were digits. If you controlled the pin with regular digital write, you'd not only control the colon, but also the additional dot on the top.  
Using PWM on those segments is a bad idea because you have no control over when both anode and cathode are enabled - theoretically, if the library ran at just the right speed, the segment's won't light up until some high percentage of PWM because the *enabled* time of the PWM could be synced with the *disabled* time of those particular common lines.

basgoossen, 2023-01-15:
I did not check the connection schema carefully enough. My display has a separate anode and cathode for those led's (and is hence using 14 pins instead of 13).
i get your point for PWM, however a higher PWM frequency (arduino supports up to 62khz) mitigates this issue.