untr0py/SevSeg #97: Flickering/Out-of-sync display when combining serial output with displaying on the 7seg

Omar-Nour, 2022-04-30:
It seems that using serial print along with the 7seg slows down the output to the display, resulting in the display being out of sync and flickering.

machmar, 2022-04-30:
That is not any issue of the library. It just means that your code is too slow, and the library can't control the display fast enough.  
  
Say you have the serial communication set to 9600 baud rate, and you're using the default settings (1 start bit; no parity; 1 stop bit). That means 9600 bits are sent per second (0.10416 ms per bit). Sending one byte (one character) takes 1ms. Sending a message like `program speed = 18` takes 18.75 ms.  
  
It is most likely that my calculations are wrong, but you get what I mean. By spending 18 ms in the Serial print function, the display can only be controlled at that slow speed. Even more, since the display is common cathode/anode, only one segment of each digit is lit up at a time. So to display all the segments, the library has to run through 8 times (total refresh speed is 6Hz).  
  
For speeding up your code, I'd suggest upping the baud rate (I usually go all the way to 2000000). If that is not possible, then use the `millis()` functionality to make the `Serial.print()` function run only once every 10ms or so. Hope this helped ☺.

machmar, 2022-04-30:
Also this has already been discussed here: https://github.com/DeanIsMe/SevSeg/issues/95

Omar-Nour, 2022-05-01:
**Thank you for the advice** and sorry for issuing an invalid problem.

closed 2022-05-01 by Omar-Nour