untr0py/SevSeg #99: Issue Displaying numbers and receiving inputs

PedroVillasBoas, 2022-11-15:
First, I think I should explain what I’m trying to do. 
 I’m making a simple “Who can press the button faster” with two arcade buttons. In a few words, 2 players will “battle” to see who can press faster a set number of presses. 
 I have 4 simple buttons, 2 arcade buttons, 1 LCD 16,2 and 1 4-Digit 7 Segment display. 

   I currently have 2 problems: 
1. When the player is at the Game Setup part, he’ll press buttons to increase the total number of presses. But if I press one time, the 4-Digit Display gets the number right away and displays it.  (I think that’s an easy fix, not my main issue)
2.  My main problem is, when the display is active (displaying the remaining number of presses of each player) none of the button work and the LCD screen stays frozen on the first number I pressed.
 I assume it’s because of the sevseg.refreshDisplay() function, but I don’t know since I started with Arduino recently. 
 The code is commented, and I tried to explain every bit of it by comment. If you have any doubts please ask, I’m desperate. :(

Here’s my code:

```
// Libraries
#include <Wire.h>               // 16,2 LCD Screen
#include <LiquidCrystal_I2C.h>  // 16,2 LCD Screen
#include <SevSeg.h>             // 4 Digit 7 Segment Display (4D7SDisplay)
///////////////// BEGIN OF MUSIC PART DECLARATION /////////////////
// Define each note to a frequence in hertz
#define NOTE_B0 31
#define NOTE_C1 33
#define NOTE_CS1 35
#define NOTE_D1 37
#define NOTE_DS1 39
#define NOTE_E1 41
#define NOTE_F1 44
#define NOTE_FS1 46
#define NOTE_G1 49
#define NOTE_GS1 52
#define NOTE_A1 55
#define NOTE_AS1 58
#define NOTE_B1 62
#define NOTE_C2 65
#define NOTE_CS2 69
#define NOTE_D2 73
#define NOTE_DS2 78
#define NOTE_E2 82
#define NOTE_F2 87
#define NOTE_FS2 93
#define NOTE_G2 98
#define NOTE_GS2 104
#define NOTE_A2 110
#define NOTE_AS2 117
#define NOTE_B2 123
#define NOTE_C3 131
#define NOTE_CS3 139
#define NOTE_D3 147
#define NOTE_DS3 156
#define NOTE_E3 165
#define NOTE_F3 175
#define NOTE_FS3 185
#define NOTE_G3 196
#define NOTE_GS3 208
#define NOTE_A3 220
#define NOTE_AS3 233
#define NOTE_B3 247
#define NOTE_C4 262
#define NOTE_CS4 277
#define NOTE_D4 294
#define NOTE_DS4 311
#define NOTE_E4 330
#define NOTE_F4 349
#define NOTE_FS4 370
#define NOTE_G4 392
#define NOTE_GS4 415
#define NOTE_A4 440
#define NOTE_AS4 466
#define NOTE_B4 494
#define NOTE_C5 523
#define NOTE_CS5 554
[... 487 lines cut ...]
      }else if (checkPlayerOneUnit == 5) {
        playerOneUnit = (0b01101101);
        sevseg.setSegmentsDigit(1, playerOneUnit);
      }else if (checkPlayerOneUnit == 6) {
        playerOneUnit = (0b01111101);
        sevseg.setSegmentsDigit(1, playerOneUnit);
      }else if (checkPlayerOneUnit == 7) {
        playerOneUnit = (0b00000111);
        sevseg.setSegmentsDigit(1, playerOneUnit);
      }else if (checkPlayerOneUnit == 8) {
        playerOneUnit = (0b01111111);
        sevseg.setSegmentsDigit(1, playerOneUnit);
      }else if (checkPlayerOneUnit == 9) {
        playerOneUnit = (0b01101111);
        sevseg.setSegmentsDigit(1, playerOneUnit);
      }
    }
    sevseg.refreshDisplay(); // refresh the display
  }
}

void DisplayPlayerTwo() {
  // Here is where the player two code
}
```

machmar, 2022-11-15:
I tried to understand your code fully but I failed. The only thing I can see is a strange-looking while loop that could vause the issue as it traps the code in a cycle.
  Here on line *503* `while (playerOnePress != 0 || playerTwoPress != 0) {  // Check to see if there's any presses remaining`. I'd start debugging from there and see what causes it.
   
 Also the way you read the buttons with the timed loop might not register shorter presses (below 150ms) but those are humanly difficult to do so I wouldn't bother with that.
  
  Hope this helps you out!

PedroVillasBoas, 2022-11-15:
I'm so sorry if the code is confusing. :( 
 I have a repository here on github with the READ ME file explaining what I want to achieve with the project. I'll paste it here so you can (hopefully) understand it better:

README File: 

```
Jarvis-Play-Mode
Just some for fun project

Description
A quick overview of the project.

What this project do?
You and a friend can battle to see who among you can press a button faster!
(It's simple, I know! But it's also fun!)

How does it work?
What will the code do? (Bit by Bit)

The Title Screen
The first thing you'll see when everything is set up is the Tittle Screen.
The Tittle Screen will display "Jarvis Play Mode" wich is the name of the game.
Bellow the tittle will the "Press Blue" phrase, so you can press the button and the game will move to the Setup Screen.

The Setup Screen
The Setup Screen is where you and your foe will define how many presses both of you will have to do to win the game.
You can press the Green Button (In my case) to increase by 1 the total number of presses.
Or maybe you think that's too slow, right? Me too! So I added a second button, the Yellow Button (Also in my case) to increase the total number of presses by 10.
Oh no! Made mistake? Don't worry, I got you covered! Press the Red Button (Also in my... You get the point) to erase the total number of presses and go back to 0.
Once both of you come to a agreement and decide how many presses you'll do, press the Blue Button to confirm it and move to the Game Start phase.

The Game Start Phase
Okay, so both of decided how many presses you'll have to do and are all set to BATTLE, isn't? GOOD!
After pressing the Blue Button on the Setup Screen a CountDown will occur (Just to give you and your opponent time to prepare). Starting from 3 until it vanishes.
Once the CountDown is complete the word "BATTLE" surrounded by swords will show up on the LCD screen and you're good to go! PRESS PRESS AND PRESS!
The 4-Digit-7-Segment-Display will show on the first 2 digits the total number of presses remaining for the player 1 and the last 2 for the player 2 (Quite obvious, right?)
Every press you and your opponent do will decrease and change the number on the Display.
When you or your foe gets to 0 it'll go to the End Screen.

The End Screen
The End Screen is not yet implemented. So... Wait for updates!
```

The line 503 is just a check to see if the players have pressed the entire quantity of wich they selected. 
Exemple: The players selected 50 presses. When the game begin, they'll have to press the arcade button 50 times as fast as possible. When one of them press the button 50 times, that line 503, the check, will stop the 4-Digit-7-Segment Display on displaying the remaining number of presses. I don't know if I made myself clearer or just confused you a little bit more. :(

My issue is: When the 4-Digit-7-Segment Display is on, none of the buttons work. They don't send any input while the Display is on. If the display is not on, not showing the remaining number of total presses, all buttons work as intended. As soon as the Display goes on, showing the remaining number of presses, the buttons stop working. 
 My guess (A shot in the dark) is the function `sevseg.refreshDisplay()` is doing something to stop the code from receive any input. But I don't know if that's it or if it's something else. And if it is the function, how would I go around that? 

Thank you anyway!!!

PedroVillasBoas, 2022-11-15:
You were right! I changed a few things on the code and now it's fully functional! Thank you!

closed 2022-11-15 by PedroVillasBoas