untr0py/SevSeg #101: Serial Error in ESP32?

JGomezC145, 2022-12-23:
Hi, I'm using the library in an esp32, but I have a strange error that I don't know if it only happens to me. 

When I enable Serial on the board, and then open the serial monitor, everything works fine until I get to the part where the library function is called, both for sevseg.begin(), and for refreshDisplay(); and the serial monitor starts showing weird symbols and not the expected text to print.

There is an image of the serial monitor:

[image]

machmar, 2022-12-23:
Hi, could you please send your code?

JGomezC145, 2022-12-25:
```cpp
#include <WiFi.h>
#include "time.h"
#include "sntp.h"
#include <SevSeg.h>
#include "notes.h"
#include "titanic.h"

SevSeg sevseg;

int opp = 0;
bool upopp = true;

int opp2 = 330;
bool upopp2 = true;

int opp3 = 410;
bool upopp3 = true;

const char* ssid = "-----";
const char* password = ""-----;

const char* ntpServer1 = "pool.ntp.org"; 
const char* ntpServer2 = "time.nist.gov"; 
const long gmtOffset_sec = -25200; 
const int daylightOffset_sec = 3600; 
const int buzzerPin = 23; 
bool milTime = false; 
bool showled = true;
int lastButtonState; 
int currentButtonState; 
int i_lectura;
int i_limite = 50;
int i_lectura2;

bool buzzed = false;

char tmHour[3];
char tmMinute[3];

#define led_Encender 26
#define led_WiFi 14
#define led_Error 33

#define arbol_rojo 25
#define arbol_star 13
#define arbol_azul 12
#define arbol_verde 27

#define arbol_rojoCh 3
#define arbol_starCh 1
#define arbol_azulCh 5
#define arbol_verdeCh 7

#define buzzCh 4

#define arbolInte 35

int WifiWaitCount = 0;

bool isError = false;
bool arbolLuz = true;

char secss[10];
char minnn[10];

char sTem[10];

String genn = "";

[... 251 lines cut ...]
    digitalWrite(arbol_verde, !digitalRead(arbol_verde)); */
  }
  sevseg.refreshDisplay();  
  printLocalTime();  // it will take some time to sync time :)

  
  //lastButtonState    = currentButtonState;      // save the last state
  //currentButtonState = digitalRead(buttonPin);

  /* if (!buzzed) {
    ledcWriteTone(buzzCh, 764);
    delay(1500);
    ledcWriteTone(buzzCh, 0);
    buzzed = !buzzed;
  } */
  /* if(lastButtonState == HIGH && currentButtonState == LOW) {
    Serial.println("The button is pressed");

    // toggle state of LED
    milTime = !milTime;

    
  } */

}```

machmar, 2022-12-25:
As I suspected, you're using Serial pins for the display. Pins 1 and 3 on the ESP are used for USART0 and therefore you need to leave them disconnected if you want to use serial communication (at least I think, I have never even seen an esp32 in my life, I know, I live under a rock). Try changing pins 1 and 3 to some other pins (in `setup()` `byte digitPins[] = {21, 3, 1, 22};`).
  
Hope This helps ☺

JGomezC145, 2022-12-26:
Yep, it works, is my first time using an esp32 too, thanks for help!!!

closed 2022-12-26 by JGomezC145