untr0py/SevSeg #102: Ghosting when using setChars

basgoossen, 2023-01-15:
I have a problem with digit practically all digits ghosting into the most significant digit when using setChars. I had issues with ghosting before when using setNum but this disappeared with a reset. However setChars consitently has this issue:

    void loop() {

        long ms = millis();
      
        sevseg.refreshDisplay();
      
        if(ms - lastMs > 1000){
          if(error > 0){
            sevseg.setNumber(error);
          } else {
            DateTime now = rtc.now();
            int hour = now.hour();
            int minute = now.minute();
            snprintf(dtime, 5, "%d%d", hour, minute);
            sevseg.setChars(dtime);
            sevseg.refreshDisplay();
            digitalWrite(SEG_SP_A, !digitalRead(SEG_SP_A));
            Serial.print("Current RTC Time: ");
            sevseg.refreshDisplay();
            Serial.print(now.unixtime());
            sevseg.refreshDisplay();
            Serial.print(" (");
            sevseg.refreshDisplay();
            Serial.print(dtime);
            sevseg.refreshDisplay();
            Serial.println(")");
            sevseg.refreshDisplay();
          }
          lastMs = ms;
        }
      }

All other info is displayed correcly on the segments.

machmar, 2023-01-15:
I'm not really sure what could be doing that, but try playing with `sevseg.setBrightness(90);` (try changing the number to some positive or negative value).  
Let us know if it helped!

basgoossen, 2023-01-15:
I did check positive brightness values, (between 5 and 100) but that did not make a difference, did not know that negative values were a thing in that function. Currently i'm running the code with a simple custom function that set's the segments according to the time. But i'll try later on to see if negative values make a difference. I like using SevSeg instead of the custom code since it is broadly used and it reduces the amount of local code that needs to be maintained.

machmar, 2023-01-15:
Yeah, for sure. One thought I have, try replacing the `snprintf` function with just a simple `strcpy(dtime, "abcd");` and see if that gets rid of the issue.

basgoossen, 2023-01-15:
Thanks for the support, replacing the input of setChars to something fixed or even now again using setNumber does not change the issue. I've now checked negative numbers in brightness and also tried toggeling resistorsOnSegments (i've got them on the digits), but this also does not make a difference.

machmar, 2023-01-15:
One thought, maybe try commenting out the `rtc.now();`.  
Also, one other thought I had, try just disabling the display all together by commenting out the refresh command. Then see if there is still some activity on the display. Might be something else messing with your pins?

basgoossen, 2023-01-15:
If something is messing with the pins it should be messing with pin 3 of the arduino nano i use as a display controller. But if i comment out sevseg and run my own little routine:

    void refreshDisplay() {
      unsigned long mt = micros();
      unsigned long d = mt - segmentOn;
    
      if(d >= 1000){ //Write a new segment every MS
        segmentOn = mt;
    
        //Next digit (wrap around on number of digits)
        dDigit = (dDigit + 1) % numDigits;
    
        //Enable active digit
        for( int i = 0 ; i < numDigits ; i++ ){
          digitalWrite(digitPins[i], i != dDigit);
        }
    
        //Fetch active segments from segmentMap
        byte segments = segmentMap[dval[dDigit]];
    
        //Keep a count of active segments
        byte onCount = 0;
        //Set segments according to segments value.
        for( int i = 0 ; i < 7 ; i++ ){
          bool on = ((segments >> i) & 0x01) == 1;
          digitalWrite(segmentPins[i], on);
          if(on) onCount++;
        }
    
        //Calculate on time based on brightness (between 1 and 100)
        digitOnTime = 10 * brightness;
    
        //Correct difference in brightness due to reduced led current when driving more segments.
        if(!resistorsOnSegments) digitOnTime = (digitOnTime / 8) * onCount;
    
        //Wait on time (controls brightness)
        delayMicroseconds(digitOnTime);
        
        //Turn off segments and digits
        for( int i = 0 ; i < numDigits ; i++ ){
          digitalWrite(digitPins[i], HIGH);
        }
        for( int i = 0 ; i < 7 ; i++ ){
          digitalWrite(segmentPins[i], LOW);
        }
      }
    }

It does not ghost...

All other code remains the same. So it seems to be happening somewhere in the way SevSeg controls the digits.

machmar, 2023-01-15:
That's tough. Could you send your whole project?

basgoossen, 2023-01-15:
Yeah no problem, how can i transfer the file to you?

machmar, 2023-01-15:
Would be easiest if you could just put it onto Pastebin and linked it here. That way, others can help as well.